almonds0166/6302view #1: Problems finding /dev/ttyUSB0

frohro, 2022-11-24:
Hi,
Thanks for the 6302 View.  I used it is my Control class at WWU a couple of years ago.  I'm getting ready to go with a new project for this spring quarter, and found bit rot has set in and it cannot seem to find my serial port.  Here is what the messages say.

`Welcome to 6302View Configuration.
Choose and Enter the number of the Microcontroller you're using:
(0): Teensy 3.2
(1): Arduino Uno (FTDI chipset)
(2): ESP8266 D1 Mini Pro (with CH340 Adapter)
(3): ESP32 Dev Module Rev C(Si Labs Device)

3

Enter the Websocket Port you'd like to use for Python to Browser Communication. Valid choices are 6300-6400.
6302

Verbose mode? (Warning may slow down system when operating at high data rates). Enter 1 for True, 0 for False
1
Ah running an ESP32. Good taste.
Starting...
/home/frohro/Classes/Feedback/Selfbalancing Car/6302/gui/./local_server.py:272: DeprecationWarning: There is no current event loop
  asyncio.get_event_loop().run_until_complete(server)
/home/frohro/Classes/Feedback/Selfbalancing Car/6302/gui/./local_server.py:273: DeprecationWarning: There is no current event loop
  asyncio.get_event_loop().run_forever()
0:   /dev/ttyS5 - n/a (Vendor ID: None)
None 4292
1:   /dev/ttyS4 - n/a (Vendor ID: None)
None 4292
2:   /dev/ttyUSB0 - USB Serial (Vendor ID: 6790)
6790 4292
USB Port: False
c
0:   /dev/ttyS5 - n/a (Vendor ID: None)
None 4292
1:   /dev/ttyS4 - n/a (Vendor ID: None)
None 4292
2:   /dev/ttyUSB0 - USB Serial (Vendor ID: 6790)
6790 4292
USB Port: False
failing on read
0:   /dev/ttyS5 - n/a (Vendor ID: None)
None 4292
1:   /dev/ttyS4 - n/a (Vendor ID: None)
None 4292
2:   /dev/ttyUSB0 - USB Serial (Vendor ID: 6790)
6790 4292
USB Port: False
failing on write
`
Any ideas?

Thanks,

Rob

almonds0166, 2022-11-24:
Hi Rob, thanks for passing along the info.  That's so nice to hear you're finding use out of 6302view (:  I've been thinking about the project recently and figured bit rot would manifest soon.

Is this macOS or a Linux distro?  And is this Python 3.10 or 3.11? -- it looks like `asyncio.get_event_loop` was [recently deprecated](https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.get_event_loop), but that doesn't seem to be the issue here.

I'll look into this as soon as I can.

Cheers
-- Maddie

frohro, 2022-11-24:
Thanks, Maddie, I'm using Ubuntu 22.04.1 LTS with Python 3.10.6.  I also tried using a brand new install of Ubuntu 22.10, and python 3.10.7; it is the same.  The errors occur when you start or refresh the page.  Let me know what testing I can do for you as you look into this.  Thanks so much for giving your time to help us all out.  My students liked this software the last time I taught the class, so I wanted to use it again.  It is better than using the plot window of the Arduino IDE, and we are controlling some Keystudio self-balancing cars this year, and I'm hoping to also use the websocket option without a tether.   Last year it was unnecessary, but this year it would be nice.  I'll try it out now, as it doesn't need the USB port, and see what happens. :-)
Again, thanks!
Rob

almonds0166, 2022-11-25:
I've been playing around with it on my Linux Mint 20.2, and I haven't ran into this issue -- I might have an Ubuntu flash somewhere...  I've been testing with Python v3.10.8 as well as with v3.8.10.  Refreshing the page or clicking Reconnect works as I'd expect.

`pip freeze` gives me the following for both of my virtual environments:

```
pyserial==3.5
websockets=10.4
```

I'm assuming your serial and websocket modules are up to date too?  And can you list your ports, from pyserial's perspective?

```python -m serial.tools.list_ports```

On my end I get 1 port found as expected, `/dev/ttyUSB0`.

I was going to ask what sketch you have flashed to your ESP32, although that shouldn't matter for detecting `/dev/ttyUSB0`, and my `local_server.py` can detect the device even with a blank `setup()` and `loop()` loaded.

jodalyst, 2022-11-27:
I probably won't be able to help with this for a few weeks, (or maybe next weekend at the earliest). Let me know what you all figure out in the mean time.

almonds0166, 2022-12-08:
Hey Rob, did you happen to test around a bit? Although I'm finishing up some grad apps this and next week, I expect to have time to set up or find an Ubuntu distribution to test this on more deservedly. I've also updated the `asyncio` code to work with Python v3.10 (and continue to work with previous versions) on my local code, although I'm still tinkering with it because it no longer has a graceful shutdown with Ctrl+C.

Cheers

almonds0166, 2023-02-03:
This has been on my mind at least twice per week for the past two months. I just pushed an update for the `local_server.py` code, which works on my Linux Mint, macOS, and Windows.

Next week, I'm making an effort to find an Ubuntu distro I can test this on. And I plan to give the Python some more polish soon, since 90% of the time I hit Ctrl+C, the server shuts down not gracefully, and the script could use some reorganization.

frohro, 2023-02-03:
Thanks so much for your work on this.  I haven't done anything recently, as school has me busy.  I do hope to test things out before the next quarter starts at the end of March.

Thanks so much!

Rob

On 2/3/23 1:28 PM, almonds0166 wrote:
CAUTION: This email originated from outside the Walla Walla University email system.

This has been on my mind at least twice per week for the past two months. I just pushed an update for the local_server.py code, which works on my Linux Mint, macOS, and Windows.

Next week, I'm making an effort to find an Ubuntu distro I can test this on. And I plan to give the Python some more polish soon, since 90% of the time I hit Ctrl+C, the server shuts down not gracefully, and the script could use some reorganization.

—

Rob Frohne,PhD PE
Professor
School of Engineering
Walla Walla University
204 S. College Ave.
College Place, WA 99324

Excellence in Thought • Generosity in Service • Beauty in Expression • Faith in God

almonds0166, 2023-02-07:
I was chatting with a friend about this and they pointed out:

```
...
0: /dev/ttyS5 - n/a (Vendor ID: None)
None 4292
1: /dev/ttyS4 - n/a (Vendor ID: None)
None 4292
2: /dev/ttyUSB0 - USB Serial (Vendor ID: 6790)
6790 4292
...
```

Your `/dev/ttyUSB0` with vendor ID `6790` is an ESP8266, not an ESP32. 😄 Apparently a similar situation happened to my friend in 6.111 long ago, lucky for us.

I can ~~'t~~ believe I overlooked that OTL ... This gives me ideas on how to improve the script for the next update.

Can you confirm that this was the case, and I'll close the issue?

Cheers (:

almonds0166, 2023-02-21:
Pushed some updates to the `local_server.py` that I'm proud to announce!
* cleaned up the script, so it's easier to scan and understand
* added color for glanceability (I tested it, and it works on Linux Mint, Windows, and macOS)
* smarter port finding
* smarter error handling

Next I'd like to look into refining/updating the JavaScript in the future sometime.

Let us know if any other bugs come up.  Thank you for your patience!

closed 2023-02-21 by almonds0166