almonds0166/6302view #3: Toggle Does not Initialize Correctly if Initially the bool is False

frohro, 2023-05-08:
When I create a toggle, I have to actually toggle it to have its value be correct if it starts out as false.  If it is true, all is well, but sometimes I want it to start as false.    Here is some code that shows the behavior:

`
#include <Six302.h>

/* For this demo, make sure to have the `#define S302_SERIAL`
   macro enabled in the library. 

   This demo presents a button that when pressed increments
   the number displayed */

// microseconds
#define STEP_TIME 10000
#define REPORT_TIME 50000

CommManager cm(STEP_TIME, REPORT_TIME);

bool input = false;
int32_t output = 1;

void setup() {

  /* Add modules */
  cm.addToggle(&input, "Input Boolean");
  cm.addNumber(&output, "Boolean Value as Int");

  /* Ready to communicate over Serial */
  cm.connect(&Serial, 115200);
}

void loop() {
  output = int32_t(input);
  cm.step();
}`

Next, this might be a feature request, but it would be nice to have the Toggle stay in sync with the Arduino.  It other words, if the arduino in the loop changes the bool value, it would be nice if the toggle reflected that.  

Thanks for all your work on this.  It is being very useful for my Control class this quarter.

Rob

almonds0166, 2023-05-09:
It's nice to hear how 6302view is instrumental in your Control class! That sort of use was the initial vision. (:

The build string, which the microcontroller hands off to the GUI, does contain the initial values. I feel like I remember this being a weird bug on the JavaScript side during development years ago. I'll look into it. Maybe I can update the GUI as I've been meaning to, while I'm at it.

As for keeping toggles in sync as the Arduino changes the values, I'm not sure how possible that is, given the current underlying setup, I'll have to look into it. The Reporters (plots, numbers) report their values, but I'm not sure the Controls (toggles, buttons, sliders) do, for ROM optimization reasons.

almonds0166, 2023-05-26:
(I'm finishing up some end-of-semester work. This is still on my mind, I've already poked around a bit, and I expect to get back into it next week. :slightly_smiling_face:)

almonds0166, 2023-06-04:
I was mistaken, it turns out that no initial values are present in the build string.

I'll look into adding the initial values for Toggles to the build string. :P

almonds0166, 2023-06-04:
Good news! The initial values *are* in the build string (at the end), and I was able to find the bug, which was in the JavaScript for the Toggle's `update` function. It works like a charm now, as it should. (:

(closing)

closed 2023-06-04 by almonds0166