almonds0166/6302view #6: ESP32-C3 Super Mini Support

frohro, 2025-05-13:
Hi Maddie,
To make our [ESP32-C3 Super Mini](https://dl.artronshop.co.th/ESP32-C3%20SuperMini%20datasheet.pdf) development boards work I had to add them in to local_server.py.  Here is what worked:
`
# Supported device IDs
VALID_MCUs: typing.List[MCU] = [
    MCU({5824}, "Teensy 3.2", "Teensy", "a Teensy"),
    MCU({10755, 9025}, "Arduino Uno (FTDI chipset)", "Arduino Uno", "an Uno"),
    MCU({6790}, "ESP8266 D1 Mini Pro (with CH340 Adapter)", "ESP8266", "an ESP8266"),
    MCU({4292}, "ESP32 Dev Module", "ESP32", "an ESP32"),
    MCU({12346}, "ESP32-C3 Super Mini", "ESP32-C3", "an ESP32"),
]
`
Thanks for your useful software. I'm still using it with our Control Systems class.
Rob

almonds0166, 2025-05-13:
That's so lovely to hear, I'll patch when I get home, thanks Rob

almonds0166, 2025-05-13:
I'll lump that vendor ID into the ESP32 category, like this:

```py
VALID_MCUs: typing.List[MCU] = [
    MCU({5824}, "Teensy 3.2", "Teensy", "a Teensy"),
    MCU({10755, 9025}, "Arduino Uno (FTDI chipset)", "Arduino Uno", "an Uno"),
    MCU({6790}, "ESP8266 D1 Mini Pro (with CH340 Adapter)", "ESP8266", "an ESP8266"),
    MCU({4292, 12346}, "ESP32 Dev Module", "ESP32", "an ESP32"),
]
```

Because, while I don't have an ESP32-C3 Super Mini to test with, my **Xiao ESP32-C3** gives `12346` for the vendor ID, and so do my **ESP32-C6 DevKitC** and **-S3 DevKitC** boards. (I never got around to checking those vendor IDs till now.)

I can't find a quick way with the `pyserial` module to reliably differentiate between the -C3, -S3, -C6, etc., so I'll commit this snippet for the sake of broadest accuracy. If you'd like the script wizard to explicitly call it the ESP32-C3 Super Mini for your class, I'd recommend keeping your snippet the way it is, maybe make a zip file or fork.

I'll leave this issue open for a day or two more in case you have extra thoughts.

Cheers
/m

closed 2025-05-14 by almonds0166