almonds0166/6302view #7: Guru meditation panic using websockets

frohro, 2025-05-17:
Hi Maddy,
I am having trouble with the ESP32 and ESP32-C3 (haven't tried any others) with websockets.  Serial seems to work, but a lot of my students are balancing things that would be much better without a tether cable.  I have been trying to fix it this afternoon, but it is not fixing easily.  I wonder if you have any advice.  Here is the output.

> ets Jun  8 2016 00:22:57␍␊
␍␊
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
configsip: 0, SPIWP:0xee␍␊
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
mode:DIO, clock div:2␍␊
load:0x3fff0030,len:1184␍␊
load:0x40078000,len:13232␍␊
load:0x40080400,len:3028␍␊
entry 0x400805e4␍␊
[     3][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)␍␊
[    10][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)␍␊
[    24][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz␍␊
[    66][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)␍␊
[    76][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation␍␊
[    87][V][esp32-hal-uart.c:467] uartBegin(): UART0 initialization done.␍␊
␊
␊
Starting ESP32 WiFi PlotExample␍␊
Connecting to WiFi through CommManager...␍␊
Connecting to Frohne-2.4GHz WiFi [  1127][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY␍␊
[  1214][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0␍␊
[  1212][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Started␍␊
[  1231][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START␍␊
.[  1331][V][WiFiGeneric.cpp:356] _arduino_event_cb(): STA Connected: SSID: Frohne-2.4GHz, BSSID: 14:91:82:60:9d:c6, Channel: 11, Auth: OPEN␍␊
[  1344][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED␍␊
[  1368][V][WiFiGeneric.cpp:370] _arduino_event_cb(): STA Got New IP:192.168.1.5␍␊
[  1376][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP␍␊
[  1383][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.1.5, MASK: 255.255.255.0, GW: 192.168.1.1␍␊
 connected!␍␊
--> 192.168.1.5:80 <--␊
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.␍␊
␍␊
Core  1 register dump:␍␊
PC      : 0x00060823  PS      : 0x00060130  A0      : 0x800d2d20  A1      : 0x3ffb21d0  ␍␊
A2      : 0x00060823  A3      : 0xcab045da  A4      : 0xcab045da  A5      : 0x3ffb218c  ␍␊
A6      : 0x3ffc61c0  A7      : 0x3ffb218c  A8      : 0x800d7a76  A9      : 0x3ffb21b0  ␍␊
A10     : 0x3ffb7d90  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  ␍␊
A14     : 0x3ffcf578  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000014  ␍␊
EXCVADDR: 0x00060820  LBEG    : 0x400d6d20  LEND    : 0x400d6d32  LCOUNT  : 0x00000000  ␍␊
␍␊
␍␊
Backtrace: 0x00060820:0x3ffb21d0 0x400d2d1d:0x3ffb21f0 0x400d2bf9:0x3ffb2240 0x400db072:0x3ffb2290␍␊
␍␊
␍␊
␍␊
␍␊
ELF file SHA256: d0f8999387fba1aa␍␊
␍␊
Rebooting...␍␊

As you can see, it happens just after the cm.connect connects to WiFi.  I have been using the ESP32 dev kit for this, because it is better tested than the ESP32-C3 in general.  My code is in the guru-meditition branch of this repo on github.  https://github.com/frohro/WWU6302view.git
If you have any good ideas, I would be grateful!
Rob

almonds0166, 2025-05-17:
Hm... did you uncomment `#define S302_WEBSOCKETS` in `Six302.h`? Asking because it doesn't look like it in the branch you shared

I can test it tomorrow morning on my hardware

frohro, 2025-05-17:
> CAUTION: This email originated from outside the Walla Walla University email system.

Yes, I did that.  I found it better to comment both ways out and change it with a -DSix302_WEBSOCKETS, but yes, that was the first thing.  🙂  I'll be interested to see if your superior programming skills will be the ticket.  🙂
Thanks,
Rob

almonds0166, 2025-05-17:
I confirmed it crashes after connecting on my little C3, and I figured out it's due to the [WebSockets](https://github.com/Links2004/arduinoWebSockets) library, which has had some updates over the months. 6302view works until WebSockets [version 2.4.1](https://github.com/Links2004/arduinoWebSockets/releases/tag/2.4.1) (June 2024) it seems.

**Version 2.4.0** is the latest for which it does not crash for me, if you need a quick fix. I will look into what can be done on the 6302view side of things.

Careful using so much Claude generated code :sweat_smile: it looks like it doesn't know what the `CommManager` parameters mean, delays are not necessary when using `cm.step()`, and I'm not sure what [the modifications to the .cpp](https://github.com/frohro/WWU6302view/blob/guru-meditation/src/Six302.cpp#L130) are trying to accomplish.

frohro, 2025-05-17:
Thanks Maddy!  I'll just use websockets 2.4.1.  That is a fine solution for me at the moment.  And yes, after the coding was above my level of expertise, I went to Claude.  I'm trying to learn how to use it in a good way.  It has been helpful for me, but you need to learn how to use it properly, which takes time.  I did learn a few things from the AI, and as you noticed, I found it confusing your CommManager with someone else's it knew about already.  The reason the Six302.cpp and .h were in the src directory is that GitHub Copilot can read the code in src .  I started to put Websockets in src too, but it was too big and I quickly abandoned that plan.  I should have done as you did and tried some old Websocket libraries.  I won't be using the modified code here.   The only modifications I did were to put const in front of char in the call to CommManager::connect(const char*, const char*) and not defining either S302_WEBSOCKETS or S302_SERIAL in the Six302.h, but instead defining them in the platformio.ini.  That way, I don't have to modify your library, only how I build it.  
Again, thanks so much for the help!!
Rob